NVlabs/RVT #2: This error occurs when I run eval.py

chenjiayi-zxx, 2023-07-04:
I wanted to run eval.py, but this error occurred.

2023-07-04 15:30:38.419021: I tensorflow/core/util/port.cc:110] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2023-07-04 15:30:38.465609: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-04 15:30:39.192211: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
MVT Vars: {'training': True, '_parameters': OrderedDict(), '_buffers': OrderedDict(), '_non_persistent_buffers_set': set(), '_backward_pre_hooks': OrderedDict(), '_backward_hooks': OrderedDict(), '_is_full_backward_hook': None, '_forward_hooks': OrderedDict(), '_forward_hooks_with_kwargs': OrderedDict(), '_forward_pre_hooks': OrderedDict(), '_forward_pre_hooks_with_kwargs': OrderedDict(), '_state_dict_hooks': OrderedDict(), '_state_dict_pre_hooks': OrderedDict(), '_load_state_dict_pre_hooks': OrderedDict(), '_load_state_dict_post_hooks': OrderedDict(), '_modules': OrderedDict(), 'depth': 8, 'img_feat_dim': 3, 'img_size': 220, 'add_proprio': True, 'proprio_dim': 4, 'add_lang': True, 'lang_dim': 512, 'lang_len': 77, 'im_channels': 64, 'img_patch_size': 11, 'final_dim': 64, 'attn_dropout': 0.1, 'decoder_dropout': 0.0, 'self_cross_ver': 1, 'add_corr': True, 'add_pixel_loc': True, 'add_depth': True, 'pe_fix': True}
Agent Information
<rvt.models.rvt_agent.RVTAgent object at 0x7f81b1d98100>
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/data/cjy/CoppeliaSim_Edu_V4_1_0_Ubuntu20_04" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, webgl, xcb.

chenjiayi-zxx, 2023-07-04:
It should be because I can't use the graphical interface on the server, is there any option to evaluate it without qt.

imankgoyal, 2023-07-04:
Hi,

Thanks for reporting this issue. Yes, from my experinece, it could be because you are trying to evaluate on a headless server on RLBench. Unfortunately, I don't have a very concrete fix for this issue. But, a couple of things that might be helpful (in order)

-- Try to use the `--headless` flg with `eval.py`.

-- Try this (fix)[https://github.com/NVlabs/RVT#gotchas].

-- Try using `xvfb-run` as discussed [here](https://github.com/stepjam/RLBench/issues/112).

-- Try using VirtualGL as discussed [here](https://github.com/stepjam/RLBench#running-headless).

Please, let me know what solution works for you.

imankgoyal, 2023-07-09:
@chenjiayi-zxx Were you able to solve this issue? It would be nice to if you can share what solution worked for you.

imankgoyal, 2023-07-21:
Closing because of inactivity. Please feel free to reopen if the issue persists.

closed 2023-07-21 by imankgoyal

LemonWade, 2023-08-11:
After downloading `VirtualGL`, using `xvfb-run` works for me

CA-TT-AC, 2024-08-27:
It works! Thanks for your sharing!

> After downloading `VirtualGL`, using `xvfb-run` works for me

LogSSim, 2024-09-14:
'''
t.qpa.plugin: Could not find the Qt platform plugin "xcb" in "code/act3d-chained-diffuser/PyRep/CoppeliaSim_Edu_V4_7_0_rev4_Ubuntu20_04"
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
'''

I tried all of these methods, but i still got this problem

LogSSim, 2024-09-15:
Oh, I solved this problem. I changed other CoppeliaSim version. It works now.

pussinmood, 2024-09-22:
I inevatiblly bumped into this issue ,and can't solve it through your previous advices,my cloud server has no x config ,is that problem? 
Hope you give me some advices.

> Hi,
>
> Thanks for reporting this issue. Yes, from my experinece, it could be because you are trying to evaluate on a headless server on RLBench. Unfortunately, I don't have a very concrete fix for [...]

alexchen09, 2025-06-16:
> Oh, I solved this problem. I changed other CoppeliaSim version. It works now.

I'm an HPC user without sudo permission so previous methods don't work for me. Would you like to share the CoppeliaSim version you used to deal with this error? Thank you so much!!