NVlabs/RVT #4: Pre-generated Replay Buffer Download

doublelei, 2023-07-05:
Hi,

Thanks for your great work. I'm trying to download your provided [pre-generated replay buffer (100 GB)](https://drive.google.com/file/d/1wOkLk8ymsp3TCFWOPOQLZZJ4OIZXRUjw/view?usp=sharing), but it seems too large and I continuously receive network error when downloading. Can you split this file to multiple zip files by task like what [PerAct](https://github.com/peract/peract#download) does? Or can you provide a script to generate the replay buffer? Thx

imankgoyal, 2023-07-05:
Great suggestion. I will do so and update.

imankgoyal, 2023-07-05:
@doublelei Here is the [link](https://drive.google.com/drive/folders/1n_vBXEL2lWmJTNxwQIuI_NinAGGhby5m?usp=drive_link) for the replay samples, broken down into individual files taskwise. Note that you only need `replay_train` for training (replay buffer is not needed for evaluation). Also, when these pregenerated replay buffer is not present, the code should generate it automatically.

imankgoyal, 2023-07-06:
Closing for now. Feel free to reopen if there are lingering issues.

closed 2023-07-06 by imankgoyal