NVlabs/RVT #5: About Video Visualization

doublelei, 2023-07-17:
Hi,

Thank you for the great video. In your demonstrations I have noted that you show the simulation video while I don't find how to save or visualize the result in the eval.py. Can you provide any instruction about how to do it? (I assume it would be like setting cinematic_recorder.enabled=True as [PerAct](https://github.com/peract/peract#recording-videos) does, but i don't find it in this codebase)

Thank you in advance.

imankgoyal, 2023-07-18:
@doublelei  thanks for your interest in our work.

This is a nice suggestion. 

One way to visualize the result in `eval.py` to run it without the `--headless` mode flag. In that way you can see the robot in action in the CoppeliaSim GUI.

Also, I just added a `--save-video` flag for `eval.py` (in 88300e55e9516aa0c40d14e6fcfbb5e316b33d72) which should also save the videos at location `<model_path>/eval/<log-name>/<model-name>/videos/`.

imankgoyal, 2023-07-21:
Closing because of inactivity. Please feel free to reopen if the issue persists.

closed 2023-07-21 by imankgoyal