NVlabs/RVT #7: How much memory is required on the GPU?

JayKamat99, 2023-08-05:
Hi, 

I am running the code on a 4GB NVIDIA RTX 3050 GPU. I am getting a runtime error while running the eval.py for RVT
`python eval.py --model-folder runs/rvt  --eval-datafolder ./data/test --tasks close_jar --eval-episodes 25 --log-name test/1 --device 0 --headless --model-name model_14.pth`

Error:
`RuntimeError: CUDA out of memory. Tried to allocate 610.00 MiB (GPU 0; 3.81 GiB total capacity; 1.77 GiB already allocated; 531.69 MiB free; 1.82 GiB reserved in total by PyTorch) If reserved memory is >> allocated memory try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF`

Is there a way I can avoid it? 
Setting max_split_size_mb does not help. I tried `PYTORCH_CUDA_ALLOC_CONF=max_split_size_mb:512` before running the code. But I am getting the same error.

imankgoyal, 2023-08-07:
Hi @JayKamat99,

For evaluation, I would recommend a GPU with at-least 16GB of memory. We used V100 with 16GB of memory.

JayKamat99, 2023-08-08:
Thanks @imankgoyal,
It works with 16GB of memory. It almost required all of it.

closed 2023-08-08 by JayKamat99

MarcNi02, 2024-12-02:
Hi @JayKamat99, 
do you know if the required memory exceeded 12GB?