NVlabs/RVT #10: paper question

LemonWade, 2023-08-09:
Thank you very much for the work you've done. May I ask a question? Can I interpret your work as being based on multi-view? I'm curious about the primary difference between your approach and multi-view. If someone were to emulate your work using just multi-view for their experiments, would they outperform you? 

Sorry for the disturbance and thank you in advance.

LemonWade, 2023-08-10:
This is exactly the answer I was looking for. Thank you!

FinnJob, 2023-08-10:
Thank you for your excellent work! 
I would like to inquire about how the occlusion problem is addressed when there is only one RGBD sensor in the system.
Thanks!

imankgoyal, 2023-08-10:
Hi @FinnJob,

Thanks for the kind words. We didn't face any occlusion issue on the tasks we tested on. I suppose with more cluttered scenes this could be an issue, but the framework is flexible enough to allow for adding an additional camera if need be.

FinnJob, 2023-08-11:
> Hi [@FinnJob](https://github.com/FinnJob),
> 
> Thanks for the kind words. We didn't face any occlusion issues on the tasks we tested on. I suppose with more cluttered scenes this could be an issue, but the framework is flexible enough to allow adding an additional camera if need be.

Thanks, it helps a lot!

closed 2023-08-11 by imankgoyal