NVlabs/RVT #20: proprioception dimension

Harimus, 2023-09-11:
Hi first of all thank you for the amazing work of RVT.

I have a question. I'm trying to reproduce some of the results on the real-robot setting with a Franka robot, and I'm trying to understand the proprioception data. According to the MTV config the `proprioception_dim` is 4.  What does those 4dim represent? If it's the joint position I assume it should bigger, and if it was cartesian position of the end-effector I assume it is either 3 or 6 (or 7 if quaternion).

Thank you in advance

imankgoyal, 2023-09-12:
Hi @Harimus,

I think #19 should address your question. Let me know if any questions still remain.

Best,
Ankit

Harimus, 2023-09-12:
ah wow I missed that one somehow, thank you!

closed 2023-09-12 by Harimus