NVlabs/RVT #21: Does the line 150 give a Cartesian coordinate of a pixel?

StarCycle, 2023-09-11:
https://github.com/NVlabs/RVT/blob/0b170d7f1e27a13299a5a06134eeb9f53d494e54/rvt/mvt/mvt_single.py#L146-L157

It seems that lines 153 and 156 are giving Cartesian coordinates of a pixel (maybe x and y). However, line 150 does not give anything because it is only related to num_images.

Is it a bug?

imankgoyal, 2023-09-12:
Hi @StarCycle,

It is not a bug. That cooridnates tells the network to which image a particular pixel belong to. For example, it tells whether a pixel belong to the front image or the top image since the images are arranged in a fixed order along the `self.num_img` dimension. Hence, I feel this information could be useful although it could be the case that the network performs fine even without it.

Best,
Ankit

StarCycle, 2023-09-12:
Thank you Ankit!

Oh I get it. It could be useful! But it is still different from the original paper, which says 

"Specifically, for each view, we render three image maps with a total of 7 channels: (1) RGB (3 channels), (2) depth (1 channel), and (3) (x, y, z) coordinates of the points in the world frame (3 channels). "

If you use (x, y, z) coordinates, could you please let me know in which line they are calculated? self.pixel_loc[:, 1, :, :] and self.pixel_loc[:, 2, :, :] seems to be (x, y) coordinates in the pixel frame, instead of the world frame

Zhuoheng

imankgoyal, 2023-09-13:
The channels for rendering, as mentioned in the paper, are [here (RGB + D)]( https://github.com/NVlabs/RVT/blob/master/rvt/mvt/mvt.py#L66-L70) and [here (x, y, z)](https://github.com/NVlabs/RVT/blob/master/rvt/mvt/mvt.py#L112-L140).

The channels [here](https://github.com/NVlabs/RVT/blob/0b170d7f1e27a13299a5a06134eeb9f53d494e54/rvt/mvt/mvt_single.py#L146-L157) are in addition to the ones above and are meant to provide position information like in other transformers.

Hope it helps!

imankgoyal, 2023-09-21:
I am not familiar with Open3D but I guess it might be possible to render coordinates with it by feeding in the XYZ instead of RGB. For this project, we chose PyTorch3D because of past experince with it :)

imankgoyal, 2023-09-26:
Closing because of inactivity.

closed 2023-09-26 by imankgoyal