NVlabs/RVT #22: Why is time an essential part of low_dim_state?

Eralien, 2023-09-15:
Hi thank you for your impressive work.
Time seems to be an essential part of the `low_dim_state` where it confuses me. Why does it matter? If we can view most point on a trajectory (if not all point) as the starting point of a new sub-trajectory, does that mean we can scale the time dim? However it does not seems to work when time was not added as part of the `low_dim_state`, or be scaled. Any insight idea about such design?

imankgoyal, 2023-09-16:
Hi, thanks for the interest in our work.

The choice of time in `low_dim_state` has been borrowed from PerAct. Time could matter as it informs the network of the current stage the task. Although the network could figure it out from the point cloud, time would be an easier signal.

I would not expect the network to work with arbitrarily scaled time but as as mentioned above it might work without it, but we haven't tried it here.

imankgoyal, 2023-09-22:
Closing because of inactivity.

closed 2023-09-22 by imankgoyal