NVlabs/RVT #29: [Training] How to evaluate when training?

Aopolin-Lv, 2023-11-03:
Hi, RVT is an amazing work, it is efficient and effective comparing with other works. At the same time, I want to ask some question about the training procedure: 
1. Do you add a validation period when training, like evaluting on the validation dataset at the end of one epoch. And if I want to do so, how can I do? 
2. How to select the best checkpoint when training?
3. Have you tried to generate more training data to enhance the model performance?

imankgoyal, 2023-11-03:
Hi @Aopolin-Lv,

Thanks for your interest in our work.

1-2: We do not have a validation period. We select the last checkpoint for testing.
3: No, we didn’t try generating more data in this work. We assume the dataset is fixed and same as PerAct.

Best,
Ankit

imankgoyal, 2023-11-11:
Closing because of inactivity.

closed 2023-11-11 by imankgoyal