NVlabs/RVT #30: Questions about the details

JisuHann, 2023-11-16:
Hi, thanks for your splendid work!
I have some questions regarding some implementation details.

1. How much time does it take for a single rendering process? Also, I am curious of the intention of how you decided to proceed the re-rendering process.
2. Have you trained your model in less number of demonstrations in simulator? I am asking this because you've used 100 demonstrations per task in simulation, and 7-14 demonstrations per task in real-world; It is quite not realistic that real-world train better with much less number of demonstrations compared to the simulation.

Thank you!

JisuHann, 2023-11-18:
I would ask via mail.

imankgoyal, 2023-11-19:
Hi,

Thanks for your interest in our work and kind words. Pardon the delayed response as I was on vacation. Adding response here for others who might have similar questions.

1. Re-rendering is very fast and done on the fly for each sample. I don’t know the exact numbers on the top of my head but it is about comparable to  the forward+backward pass for a sample. 

Re-rendering helps performance. The motivation comes from one of my prior works “Revisiting point cloud classification with a simple and effective baseline”

2. Yes we use less demonstration in real. Even with less demonstration the model could work in simulation as shown in PerAct. To keep it simple, we used the best setting in simulation to use 100 demos. Unfortunately in the paper we didn’t try 10 demos in Simulation, but my guess would be that even there the model could work well and something to try.

3. I would suggest trying to change this flag to false (
https://github.com/NVlabs/RVT/blob/master/rvt/mvt/config.py#L29)

4. Yes it works and the reason would be likely because it has seen similar artifacts in training.

5. We used the same image related augmentation as in the the re-rendering case. But the point is that we cannot apply any 3D data augmentation (like adding noise to camera locations) as that would require re-rendering. 

Hope it helps.

Best,
Ankit

JisuHann, 2023-11-19:
Thanks a lot!

imankgoyal, 2023-12-07:
Hello,

- The low_dim_state is set to 4. I am not sure about the issue you are facing. Were you able to run evaluation with the provided pre-trained weights and code? Is the issue specific to something you were changing?

- The evaluation code is provided. What do you mean by evaluation code not being set? Were you able to get the published results with the evaluation code and pre trained weights?

- I think the loss was around 6. I will try to see if I can fetch the original loss plot.

best,
Ankit

JisuHann, 2023-12-09:
Thanks for your quick reply. 
For the evaluation questions, I would check whether bug exists.

closed 2023-12-09 by JisuHann