NVlabs/RVT #32: Does the RVT model control Franka robot in Isaac-sim?

zhangsanfeng86, 2023-11-21:
Hi ,
Thank your for release the model.
Does the RVT model control Franka robot in Isaac-sim?

imankgoyal, 2023-11-23:
Hi, 

Thanks for your interest in our work. We do not test RVT in Issac-sim in this project. However, as long as the perception is the same format, i.e. calibrated point clouds, then I don't think the simulator should be an issue.

Best,
Ankit

zhangsanfeng86, 2023-11-25:
> Hi,
> 
> Thanks for your interest in our work. We do not test RVT in Issac-sim in this project. However, as long as the perception is the same format, i.e. calibrated point clouds, then I don't think the simulator should be an issue.
> 
> Best, Ankit

-----------------------
Hi Ankit,

Thank you for your reply!
Clould you show me where to get the action format fo RVT?

Best,
Peter

imankgoyal, 2023-11-28:
Hi,

I suppose you are looking for some this: https://github.com/NVlabs/RVT/blob/master/rvt/models/rvt_agent.py#L755-L762
Essentially the output is a concatenation wpt (3D loc of gripper), rot_quat, grip and collision predictions.

Best,
Ankit