NVlabs/RVT #36: Rendered image with color noise in the background

pipixiaqishi1, 2023-12-28:
Hi,

When visualizing the rendered images, I found much color noise in the background(which should be dark ideally), different from Figure 5 in your great paper. Can the color noise be removed with a special render setting? or do you manually remove it from the rendered image for better visualization?
Looking forward to your reply~

Best,
Jerry

[image]

imankgoyal, 2024-01-06:
Hi,

Thanks for your interest in our work. I haven't observed this error so I am not sure about it. Could you describe the steps that you did?

Best,
Ankit

pipixiaqishi1, 2024-01-08:
Hi,

I visualized the rendered images here and met this error. https://github.com/NVlabs/RVT/blob/0b170d7f1e27a13299a5a06134eeb9f53d494e54/rvt/mvt/mvt.py#L231-L236

The code is as follows:
```
from torchvision import transforms
import torch.nn.functional as F
unloader = transforms.ToPILImage()
for i in range(5):
      rgb = img[0,i,3:6].  # RGB channels, i_th camera
      rgb = rgb.cpu().clone()
      rgb_img = unloader(rgb)
      rgb_img.save(f'{i}.jpg')

```

imankgoyal, 2024-01-09:
All, I see the issue. This could happen in the train mode if the img_aug is not 0.

```
    if img.dtype != np.uint8:
        if not ((img.max() <= 1) and (img.min() >= 0)):
            print(
                f"In visualization, img is not in [0, 1],"
                f" it is in [{img.min()}, {img.max()}"
                f" and will be clipped."
                f" This can happen in train mode with img_aug != 0"
            )
            img[img < 0] = 0
            img[img > 1] = 1

        img = (img * 255).astype(np.uint8)
```
Can you try along the logic of the above code and see if it works?

pipixiaqishi1, 2024-01-10:
It does work! Thanks very much for your patient answers.

closed 2024-01-10 by imankgoyal