NVlabs/RVT #39: Questions about the scene bounds and img aug in real-robot experiments

TeleeMa, 2024-03-20:
Thank you for your great work.

I am trying to run the real-robot experiments with RVT, and I have two questions:

1. Do you move the point clouds into the bound in real-robot experiments? If so, how to identify the scene bounds for the real-robot experiments like in https://github.com/NVlabs/RVT/blob/0b170d7f1e27a13299a5a06134eeb9f53d494e54/rvt/models/rvt_agent.py#L533
2. Do you utilize any augmentation for real-robot training like in https://github.com/NVlabs/RVT/blob/0b170d7f1e27a13299a5a06134eeb9f53d494e54/rvt/models/rvt_agent.py#L514

Thank you very much!

imankgoyal, 2024-03-21:
Thanks for your interest in our work.

— Yes, we move the point cloud in the scene bound. The scene bound is identified manually. We made sure all the key-points and area of interest is within the scene bound. Then the value was kept constant for all tasks.

— Yes, we use the same set of augmentations as sim in real.

Best,
Ankit

TeleeMa, 2024-03-22:
Thank you for your reply! @imankgoyal 

One more question, is the coordinate origin point of the whole action space and scene bounds determined by the center point position of the robotic arm in real-robot experiments?

imankgoyal, 2024-03-26:
Hi,

I think the coordinate origin does not matter. The scene bound represents the bound (+x, -x, +y, -y, +z, -z), which is specified regarding the coordinate origin. The same is true for the action space. 

Best,
Ankit

TeleeMa, 2024-03-28:
Thank you for your reply!

closed 2024-03-28 by TeleeMa