NVlabs/RVT #45: Question about the suitable version of xformers

hnuzhy, 2024-06-28:
Hi, thanks a lot for releasing you code of awesome works RVT and RVT-2.
When I installed RVT-2 and run `pip install -e '.[xformers]'`, I still met a bug as below:
`AttributeError: '_OpNamespace' object has no attribute '_efficient_attention_forward'`. 
This bug is related to a non-defined torch function `torch.ops.aten._efficient_attention_forward`, which is imported by `xformers`. 
I used your recommended torch version 1.12.1. And I guess this problem occurred due to the rapid updating of the lib `xformers`.
Actually, in the [setup.py](https://github.com/NVlabs/RVT/blob/master/setup.py#L49) file, it always pulls down the lateset version.
Thus, I would like to ask what is the matched version of xformers you used when running RVT-2.

imankgoyal, 2024-06-28:
Hi, @hnuzhy 

I have used this version for xformers `0.0.19+70161e5.d20230419`

```
python3 -c "import xformers; print(xformers.__version__)"
0.0.19+70161e5.d20230419
```

I hope it works at your end. Please comment if it works and I can help others try the same.

Best,
Ankit

hnuzhy, 2024-06-29:
> Hi, @hnuzhy
> 
> I have used this version for xformers `0.0.19+70161e5.d20230419`
> 
> ```
> python3 -c "import xformers; print(xformers.__version__)"
> 0.0.19+70161e5.d20230419
> ```
> 
> I hope it works for you. Please comment if it does so that it can help others who are trying the same.
> 
> Best, Ankit

OK. Thank you for your reply. 
Actually, after trying, I used the `v0.0.22`, and it worked with `PyTorch 1.12.1`.
Maybe you can modified [this line](https://github.com/NVlabs/RVT/blob/master/setup.py#L49) into 
`"xformers @ git+https://github.com/facebookresearch/xformers.git@v0.0.22#egg=xformers"` or 
`"xformers @ git+https://github.com/facebookresearch/xformers.git@v0.0.19#egg=xformers"` 
for installing a matched `xformers`.

hnuzhy, 2024-06-29:
By the way, I also met the version problem when installing `PyTorch3D`. Because its installation script also contained the `@stable` suffix, which keeps changing and may not be matched with `PyTorch 1.12.1`. Although I wanted to install RVT-2, it seems that I should also have to install the `rvt/libs/peract` locally, which needs to use the `PyTorch3D`. Would you mind show us the right version of it? I have tried [Version 0.7.1](https://github.com/facebookresearch/pytorch3d/releases/tag/v0.7.1) and [Version 0.7.2](https://github.com/facebookresearch/pytorch3d/releases/tag/v0.7.2), but they all failed.

hnuzhy, 2024-06-29:
Continue from previous post, I found a workable installation way of PyTorch3D from [here](https://github.com/facebookresearch/pytorch3d/blob/main/INSTALL.md)
```bash
$ conda install -c fvcore -c iopath -c conda-forge fvcore iopath
$ conda install -c bottler nvidiacub
$ conda install pytorch3d -c pytorch3d
```
Hope this will be useful for other followers~

imankgoyal, 2024-06-30:
Thanks for sharing this info @hnuzhy! I will add it to the Readme for others to discover.

closed 2024-06-30 by imankgoyal