NVlabs/RVT #46: Use RVT-2 without pytorch3d?

Walter0807, 2024-07-03:
The readme says that "You can skip this step if you only want to use RVT-2 as it uses our custom Point-Renderer for rendering. PyTorch3D is required for RVT." However, I found that it is still used (for example [here](https://github.com/NVlabs/RVT/blob/master/rvt/mvt/augmentation.py#L8)). Is it possible to bypass pytorch3d for RVT-2?

imankgoyal, 2024-07-03:
Hi @Walter0807, 

That's a good point. The augmentation function still uses some utils from PyTorch3D, so the current code bypasses only its rendering functionality. If you want to bypass it altogether, the util functions could be replaced by similar ones from Scipy (more info [here](https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.transform.Rotation.html)). I would be happy to merge any updates if you create one.

Best,
Ankit

Walter0807, 2024-07-13:
Thanks for your reply! I managed to install PyTorch3D for backward compatibility.

closed 2024-07-13 by Walter0807