NVlabs/RVT #50: How to calculate depth from your depth image?

jianzhou0420, 2024-08-04:
Your work is incredible! It is very helpful for me to enter this field. And, can you make it clear about how to use the depth image?
I found the depth images in you RLBench datasets have 3 channels and each channel's data differ from other two. I wonder 
- How to reconstruct the actual depth information from them.
- And what is the unit of the depth info. Meters or Millimeters? 

Thank you

imankgoyal, 2024-08-04:
Hi,

I am not sure what you mean by `3 channels and each channel's data differ from other two`.  In RVT-2, we create three views of the scene, so each view would have its own depth. There are other channels in the image, including `XYZ` correspondence, `RGB` and depth. 

Also, as far as I remember, the depths should be in m.

Best,
Ankit

imankgoyal, 2024-09-10:
@jianzhou0420 I think [this issue](https://github.com/NVlabs/RVT/issues/53) should address your questions. Please let me know if you have any more questions.

imankgoyal, 2024-10-18:
Closing because of inactivity. Feel free to reopen if the issue persists.