NVlabs/RVT #52: OMPL Planner error.

RocktimJyotiDas, 2024-08-23:
Hi,

Thank you for providing such an excellent repository. I’ve encountered an issue with the OMPL planner, and I’m hoping you might be able to assist. Below is the error message I’m receiving:

```
[simExtOMPL:warning]   OMPL: /home/marc/Documents/ompl-1.4.2-Source/src/ompl/base/src/Planner.cpp:247: SBL: Skipping invalid start state (invalid state)
[simExtOMPL:error]   OMPL: /home/marc/Documents/ompl-1.4.2-Source/src/ompl/geometric/planners/sbl/src/SBL.cpp:101: SBL: Motion planning start tree could not be initialized!
```

I have followed the installation instructions for RL-bench as outlined in several repositories, but the issue persists. If you have any insights or suggestions on how to resolve this error, your guidance would be greatly appreciated. One note, I am using Ubuntu 20.04 and its respective CoppeliaSim version.

imankgoyal, 2024-09-10:
Hi,

Thanks for your interest in our work. Unfortunately, I have not encountered this error, and it seems to be RLBench or CoppeliaSim-specific. It might be better to post this query on their forums. 

Please let me know if you find the solution, and I will be happy to cross-post it here.

Best,
Ankit

imankgoyal, 2024-10-14:
Closing because of inactivity. Feel free to open if the issue persists.

closed 2024-10-14 by imankgoyal