NVlabs/RVT #53: Question about the dimension of the rendered images

Jianshu-Hu, 2024-09-05:
Hi, thank you for your great work.
I am trying to understand the details of the implementation. I notice that in mvt, there is a flag called `add_corr`. Could you explain this flag? With `add_corr=False`, I understand that the rendered img contains 7 channels which are correpondingly pixel locations (3 dims) and RGBD images (4 dims). However, I do not understand the additional 3 channels when `add_corr=False`.

Thanks in advance for any clarification about this.

imankgoyal, 2024-09-10:
Hi,

Thanks for your interest in our work. When we use `add_corr=True`, we add another 3 dims, where we render the `xyz` values of the point clouds. Hence the number of dimensions is 10 (7 + 3). Information about rendering `xyz` is same the paper ([Check Page 4, Method Rendering](https://openreview.net/pdf?id=0hPkttoGAf))

Note that `pixel locations` is different from rendering `xyz` (which is enabled with `add_corr`). Pixel location describes the "location" of each 2D pixel, i.e., on which image the pixel is located and its spatial location within the image ([defined here](https://github.com/NVlabs/RVT/blob/ff1065e2a5489d0fa7d99d945e0965fe620e6569/rvt/mvt/mvt_single.py#L204-L215)). 

I hope it helps.

Best,
Ankit

Jianshu-Hu, 2024-09-10:
Got it! Thank you so much for your reply! 

Best, Jianshu

closed 2024-09-10 by Jianshu-Hu