NVlabs/RVT #56: When

pussinmood, 2024-09-17:
Created Dataset. Time Cost: 0.10559717814127605 minutes
Traceback (most recent call last):
  File "train.py", line 311, in <module>
    mp.spawn(experiment, args=(cmd_args, devices, port), nprocs=len(devices), join=True)
  File "/root/autodl-tmp/conda/envs/rvt-2/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 240, in spawn
    return start_processes(fn, args, nprocs, join, daemon, start_method='spawn')
  File "/root/autodl-tmp/conda/envs/rvt-2/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 198, in start_processes
    while not context.join():
  File "/root/autodl-tmp/conda/envs/rvt-2/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 160, in join
    raise ProcessRaisedException(msg, error_index, failed_process.pid)
torch.multiprocessing.spawn.ProcessRaisedException: 

-- Process 0 terminated with the following error:
Traceback (most recent call last):
  File "/root/autodl-tmp/conda/envs/rvt-2/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 69, in _wrap
    fn(i, *args)
  File "/root/autodl-tmp/RVT/rvt/train.py", line 218, in experiment
    rvt = MVT(
  File "/root/autodl-tmp/RVT/rvt/mvt/mvt.py", line 82, in __init__
    from point_renderer.rvt_renderer import RVTBoxRenderer as BoxRenderer
  File "/root/autodl-tmp/RVT/rvt/libs/point-renderer/point_renderer/rvt_renderer.py", line 4, in <module>
    from point_renderer.renderer import PointRenderer
  File "/root/autodl-tmp/RVT/rvt/libs/point-renderer/point_renderer/renderer.py", line 14, in <module>
    import point_renderer._C.render as r
ImportError: /root/autodl-tmp/RVT/rvt/libs/point-renderer/point_renderer/_C.so: undefined symbol: PyCMethod_New

pussinmood, 2024-09-17:
when i try to run python train.py --exp_cfg_path configs/rvt.yaml --device 0,it can go through this error ,what went wrong ?
Really appreciate your briliant works and sharing spirit！Hope you have time to give me some guidances!

pussinmood, 2024-09-17:
My env is built by miniconda ,cuda=11.3,pytorch=1.12.1,pytorch3d=0.7.4,xformers=0.0.13,i also tried to build it by cuda 11.1,but rvt installing didn't succeed.
Installation really screwed me well~.~,please give some advices

imankgoyal, 2024-09-18:
Hi,

Thanks for your interest in our work!

Unfortunately, I have not seen this error, so it would be tough to help. 

It is clear, though, that this is an installation issue. I suggest closely following the instructions and version recommendations. Also, for PyRenderer, you can try reinstalling and see if that helps.

Best,
Ankit

pussinmood, 2024-09-18:
Appreciate your help~.~

pussinmood, 2024-09-18:
I solved these issues {(undefined symbol: PyCMethod_New)、(NotImplementedError: No operator found for this attention: AttentionOpDispatch...)} by installing xformers==0.0.28.post1 ,pytorch==2.4.1,cuda==11.8,python=3.11.9.
All the issues are because of xformers's version! 
And i install pytorch3d from this https://anaconda.org/pytorch3d/pytorch3d/files?page=2
[linux-64/pytorch3d-0.7.8-py311_cu118_pyt241.tar.bz2]

pussinmood, 2024-09-18:
training process is going well by now,but only one 4090 is pretty slow!

pussinmood, 2024-09-18:
thanks again ,hope you making something greater and make the world better!

closed 2024-09-19 by imankgoyal