NVlabs/RVT #57: ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory

rebecajohn, 2024-09-19:
(rvt) root@dfa9a66f8a08:/RVT/rvt# python train.py --exp_cfg_path configs/rvt2.yaml --mvt_cfg_path mvt/configs/rvt2.yaml --device 0
bin /miniconda3/envs/rvt/lib/python3.8/site-packages/bitsandbytes/libbitsandbytes_cpu.so
/miniconda3/envs/rvt/lib/python3.8/site-packages/bitsandbytes/cextension.py:33: UserWarning: The installed version of bitsandbytes was compiled without GPU support. 8-bit optimizers, 8-bit multiplication, and GPU quantization are unavailable.
  warn("The installed version of bitsandbytes was compiled without GPU support. "
Traceback (most recent call last):
  File "train.py", line 24, in <module>
    import rvt.models.rvt_agent as rvt_agent
  File "/RVT/rvt/models/rvt_agent.py", line 19, in <module>
    import rvt.utils.peract_utils as peract_utils
  File "/RVT/rvt/utils/peract_utils.py", line 8, in <module>
    from rvt.models.peract_official import create_agent_our
  File "/RVT/rvt/models/peract_official.py", line 6, in <module>
    from rvt.libs.peract.agents.peract_bc.launch_utils import create_agent
  File "/RVT/rvt/libs/peract/agents/peract_bc/launch_utils.py", line 22, in <module>
    from rlbench.backend.observation import Observation
  File "/RVT/rvt/libs/RLBench/rlbench/__init__.py", line 4, in <module>
    import pyrep
  File "/RVT/rvt/libs/PyRep/pyrep/__init__.py", line 5, in <module>
    from .pyrep import PyRep
  File "/RVT/rvt/libs/PyRep/pyrep/pyrep.py", line 3, in <module>
    from pyrep.backend import sim, utils
  File "/RVT/rvt/libs/PyRep/pyrep/backend/sim.py", line 2, in <module>
    from ._sim_cffi import ffi, lib
ImportError: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory

odoyhaha, 2024-09-19:
try this
https://mega.co.nz/#!B31G2LaZ!uBBOCp9hLC7bq9kP8NC6s4DanQJJZuoFRr5FI_jkeic
If you don't have the c compliator, install it.(gcc or clang)

pussinmood, 2024-09-19:
Step4:didn't add source ~/.bashrc?

rebecajohn, 2024-09-19:
source ~/.bashrc was done.

imankgoyal, 2024-09-21:
This seems to be an installation issue with CoppeliaSim/PyRep. I would suggest following all the installation steps closely.

closed 2024-09-22 by rebecajohn

SajjadPSavoji, 2025-02-07:
> source ~/.bashrc was done.

Hi, were you able to solve this problem?
@rebecajohn