NVlabs/RVT #60: RuntimeError: The call failed on the V-REP side. Return value: -1

pussinmood, 2024-09-22:
Sorry to bother you again,but i really need your help ,after really struggling with this "qt.qpa.xcb: could not connect to display :1.0 qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.". I handled it with this[https://forum.qt.io/topic/132855/qt-qpa-xcb-could-not-connect-to-display-1-0-qt-qpa-plugin-could-not-load-the-qt-platform-plugin-xcb-in-even-though-it-was-found](url).
But new issue appear after that ,trace back is like :
Traceback (most recent call last):
  File "/home/fb/RVT/rvt/eval.py", line 556, in <module>
    _eval(args)
  File "/home/fb/RVT/rvt/eval.py", line 510, in _eval
    scores = eval(
             ^^^^^
  File "/home/fb/.conda/envs/rvt/lib/python3.11/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/eval.py", line 300, in eval
    raise e
  File "/home/fb/RVT/rvt/eval.py", line 294, in eval
    for replay_transition in generator:
  File "/home/fb/RVT/rvt/libs/YARR/yarr/utils/rollout_generator.py", line 40, in generator
    obs = env.reset_to_demo(eval_demo_seed)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/utils/custom_rlbench_env.py", line 34, in reset_to_demo
    desc, obs = self._task.reset_to_demo(d)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/RLBench/rlbench/task_environment.py", line 161, in reset_to_demo
    return self.reset()
           ^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/RLBench/rlbench/task_environment.py", line 75, in reset
    desc = self._scene.init_episode(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/RLBench/rlbench/backend/scene.py", line 136, in init_episode
    self.task.validate()
  File "/home/fb/RVT/rvt/libs/RLBench/rlbench/backend/task.py", line 274, in validate
    self._waypoints = self._get_waypoints()
                      ^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/RLBench/rlbench/backend/task.py", line 410, in _get_waypoints
    feasible, way_i = self._feasible(waypoints)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/RLBench/rlbench/backend/task.py", line 361, in _feasible
    path = point.get_path(ignore_collisions=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/RLBench/rlbench/backend/waypoints.py", line 55, in get_path
    path = arm.get_path(self._waypoint.get_position(),
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/PyRep/pyrep/robots/arms/arm.py", line 444, in get_path
    p = self.get_nonlinear_path(
        ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/PyRep/pyrep/robots/arms/arm.py", line 385, in get_nonlinear_path
    _, ret_floats, _, _ = utils.script_call(
                          ^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/PyRep/pyrep/backend/utils.py", line 65, in script_call
    return sim.simExtCallScriptFunction(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fb/RVT/rvt/libs/PyRep/pyrep/backend/sim.py", line 698, in simExtCallScriptFunction
    _check_return(ret)
  File "/home/fb/RVT/rvt/libs/PyRep/pyrep/backend/sim.py", line 27, in _check_return
    raise RuntimeError(
RuntimeError: The call failed on the V-REP side. Return value: -1

pussinmood, 2024-09-22:
And before this trace back ,lots of "QFactoryLoader::QFactoryLoader() looking at "/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/b0_bubbleRob"
"Failed to extract plugin meta data from '/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/b0_bubbleRob'"
not a plugin" appeared many many times,i have no idea what went wrong.

pussinmood, 2024-09-22:
Then i want to eval one task only ,it also appeared hundreds of  "Failed to extract plugin meta data from '/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/b0_bubbleRob'"
not a plugin"  ,and this ends with "Error: signal 11:

/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_Z11_segHandleri+0x30)[0x7f1c3dde4ae0]
/lib/x86_64-linux-gnu/libc.so.6(+0x43090)[0x7f1d9c319090]
/lib/x86_64-linux-gnu/libc.so.6(+0x188915)[0x7f1d9c45e915]
/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_Z15initGl_ifNeededv+0x11c)[0x7f1c3df8acac]
/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_ZN19COffscreenGlContextC1EiiiP9QGLWidgetii+0x1d2)[0x7f1c3dfeeff2]
/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_ZN20CVisionSensorGlStuffC1EiiibP9QGLWidgetbbii+0x7d)[0x7f1c3dff083d]
/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_ZN13CVisionSensor45createGlContextAndFboAndTextureObjectIfNeededEb+0x19d)[0x7f1c3dcb692d]
/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_ZN13CVisionSensor13detectEntity2Eibbbb+0x35f)[0x7f1c3dcbb46f]
/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_ZN13CVisionSensor44detectVisionSensorEntity_executedViaUiThreadEibbbb+0x79c)[0x7f1c3dcbbf1c]
/home/fb/RVT/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_ZN9CUiThread27__executeCommandViaUiThreadEP16SUIThreadCommandS1_+0x14a2)[0x7f1c3df437c2]".
No traceback this time...

imankgoyal, 2024-09-22:
Related issue #2 

Did you try the possible solution there?

I have found this issue to be tricky to handle, and it depends significantly on the system configuration. I recommend first trying a local system with a display. Then, if the same setup doesn't work on the remote, try the `xvfb` solution.

pussinmood, 2024-09-23:
Thanks for your patience,i will try this out.

pussinmood, 2024-09-23:
It seems my server has no X , i shoud find a way to fix it

pussinmood, 2024-09-23:
i solved these issues by virtualGL and opengl_headless, copied here ,hope to help someone like me:

apt update && apt install -y libglu1-mesa-dev mesa-utils xterm xauth x11-xkb-utils xfonts-base xkb-data libxtst6 libxv1

# install libjpeg-turbo和turbovnc
export TURBOVNC_VERSION=2.2.5
export LIBJPEG_VERSION=2.0.90
wget http://aivc.ks3-cn-beijing.ksyun.com/packages/libjpeg-turbo/libjpeg-turbo-official_${LIBJPEG_VERSION}_amd64.deb
wget http://aivc.ks3-cn-beijing.ksyun.com/packages/turbovnc/turbovnc_${TURBOVNC_VERSION}_amd64.deb
dpkg -i libjpeg-turbo-official_${LIBJPEG_VERSION}_amd64.deb
dpkg -i turbovnc_${TURBOVNC_VERSION}_amd64.deb
rm -rf *.deb

# 启动VNC服务端，这一步可能涉及vnc密码配置（注意不是实例的账户密码）。另外如果出现报错xauth未找到，那么使用apt install xauth再安装一次
rm -rf /tmp/.X1*  # 如果再次启动，删除上一次的临时文件，否则无法正常启动
USER=root /opt/TurboVNC/bin/vncserver :1 -desktop X -auth /root/.Xauthority -geometry 1920x1080 -depth 24 -rfbwait 120000 -rfbauth /root/.vnc/passwd -fp /usr/share/fonts/X11/misc/,/usr/share/fonts -rfbport 6006

# 检查是否启动，如果有vncserver的进程，证明已经启动
ps -ef | grep vnc

pussinmood, 2024-09-23:
Big gratitude，great love！

closed 2024-09-24 by imankgoyal

dongyu5656, 2024-12-22:
thank you for your work