NVlabs/RVT #61: xformers

codeshop715, 2024-10-03:
What version of xformers is used?

codeshop715, 2024-10-04:
I used xformers(version 0.0.28.post1), but the memory usage of the model has not been optimized. With 8 NVIDIA 3090, I still can’t set the batch size to 24. Here is my conda environment:
absl-py                      2.1.0
antlr4-python3-runtime       4.8
astunparse                   1.6.3
autocommand                  2.2.2
backports.tarfile            1.2.0
bitsandbytes                 0.38.1
Brotli                       1.0.9
certifi                      2024.8.30
cffi                         1.17.1
charset-normalizer           3.3.2
clip                         1.0                
colorama                     0.4.6
contourpy                    1.3.0
cycler                       0.12.1
dataclasses                  0.8
decorator                    4.4.2
einops                       0.8.0
filelock                     3.13.1
flatbuffers                  24.3.25
fonttools                    4.54.1
freetype-py                  2.5.1
fsspec                       2024.2.0
ftfy                         6.2.3
future                       0.18.3
fvcore                       0.1.5.post20221221
gast                         0.6.0
google-pasta                 0.2.0
grpcio                       1.66.2
h5py                         3.12.1
huggingface-hub              0.25.1
hydra-core                   1.0.0
idna                         3.10
imageio                      2.35.1
imageio-ffmpeg               0.5.1
importlib_metadata           8.0.0
importlib_resources          6.4.0
inflect                      7.3.1
iopath                       0.1.10
jaraco.collections           5.1.0
jaraco.context               5.3.0
jaraco.functools             4.0.1
jaraco.text                  3.12.1
Jinja2                       3.1.3
keras                        3.6.0
kiwisolver                   1.4.7
libclang                     18.1.1
Markdown                     3.7
markdown-it-py               3.0.0
MarkupSafe                   2.1.5
matplotlib                   3.9.2
mdurl                        0.1.2
ml-dtypes                    0.4.1
more-itertools               10.3.0
moviepy                      1.0.3
mpmath                       1.3.0
namex                        0.0.8
natsort                      8.4.0
networkx                     3.2.1
numpy                        1.23.5
nvidia-cublas-cu12           12.1.3.1
nvidia-cuda-cupti-cu12       12.1.105
nvidia-cuda-nvrtc-cu12       12.1.105
nvidia-cuda-runtime-cu12     12.1.105
nvidia-cudnn-cu12            9.1.0.70
nvidia-cufft-cu12            11.0.2.54
nvidia-curand-cu12           10.3.2.106
nvidia-cusolver-cu12         11.4.5.107
nvidia-cusparse-cu12         12.1.0.106
nvidia-nccl-cu12             2.20.5
nvidia-nvjitlink-cu12        12.1.105
nvidia-nvtx-cu12             12.1.105
omegaconf                    2.1.2
opencv-python                4.10.0.84
opt_einsum                   3.4.0
optree                       0.13.0
packaging                    24.1
pandas                       1.4.1
peract_colab                 0.0.0             
pillow                       10.4.0
pip                          24.2
platformdirs                 4.2.2
point_renderer               0.2.0              
portalocker                  2.10.1
proglog                      0.1.10
protobuf                     4.25.5
psutil                       6.0.0
pycparser                    2.21
pyglet                       2.0.17
Pygments                     2.18.0
PyOpenGL                     3.1.0
pyparsing                    3.1.4
pyquaternion                 0.9.9
pyrender                     0.1.45
PyRep                        4.1.0.3           
PySocks                      1.7.1
python-dateutil              2.9.0.post0
pytorch3d                    0.7.8             
pytz                         2024.2
PyYAML                       6.0.2
regex                        2024.9.11
requests                     2.32.3
rich                         13.9.1
rlbench                      1.2.0              
rvt                          0.0.1             
safetensors                  0.4.5
scipy                        1.14.1
setproctitle                 1.3.3
setuptools                   75.1.0
six                          1.16.0
sympy                        1.12
tabulate                     0.9.0
tensorboard                  2.17.1
tensorboard-data-server      0.7.2
tensorflow                   2.17.0
tensorflow-io-gcs-filesystem 0.37.1
termcolor                    2.4.0
timeout-decorator            0.5.0
tokenizers                   0.20.0
tomli                        2.0.1
torch                        2.4.1+cu121
torchaudio                   2.4.1+cu121
torchvision                  0.19.1+cu121
tqdm                         4.66.5
transformers                 4.45.1
transforms3d                 0.4.2
trimesh                      4.4.9
triton                       3.0.0
typeguard                    4.3.0
typing_extensions            4.12.2
tzdata                       2024.2
urllib3                      2.2.3
wcwidth                      0.2.13
Werkzeug                     3.0.4
wheel                        0.44.0
wrapt                        1.16.0
xformers                     0.0.28.post1
yacs                         0.1.8
yarr                         0.1               
zipp                         3.19.2

codeshop715, 2024-10-04:
I used xformers(version 0.0.28.post1), but the memory usage of the model has not been optimized. With 8 NVIDIA 3090, I still can’t set the batch size to 24. Here is my conda environment:
```
absl-py 2.1.0
antlr4-python3-runtime 4.8
astunparse 1.6.3
autocommand 2.2.2
backports.tarfile 1.2.0
bitsandbytes 0.38.1
Brotli 1.0.9
certifi 2024.8.30
cffi 1.17.1
charset-normalizer 3.3.2
clip 1.0
colorama 0.4.6
contourpy 1.3.0
cycler 0.12.1
dataclasses 0.8
decorator 4.4.2
einops 0.8.0
filelock 3.13.1
flatbuffers 24.3.25
fonttools 4.54.1
freetype-py 2.5.1
fsspec 2024.2.0
ftfy 6.2.3
future 0.18.3
fvcore 0.1.5.post20221221
gast 0.6.0
google-pasta 0.2.0
grpcio 1.66.2
h5py 3.12.1
huggingface-hub 0.25.1
hydra-core 1.0.0
idna 3.10
imageio 2.35.1
imageio-ffmpeg 0.5.1
importlib_metadata 8.0.0
importlib_resources 6.4.0
inflect 7.3.1
iopath 0.1.10
jaraco.collections 5.1.0
jaraco.context 5.3.0
jaraco.functools 4.0.1
jaraco.text 3.12.1
Jinja2 3.1.3
keras 3.6.0
kiwisolver 1.4.7
libclang 18.1.1
Markdown 3.7
markdown-it-py 3.0.0
MarkupSafe 2.1.5
matplotlib 3.9.2
mdurl 0.1.2
ml-dtypes 0.4.1
more-itertools 10.3.0
moviepy 1.0.3
mpmath 1.3.0
namex 0.0.8
natsort 8.4.0
networkx 3.2.1
numpy 1.23.5
nvidia-cublas-cu12 12.1.3.1
nvidia-cuda-cupti-cu12 12.1.105
nvidia-cuda-nvrtc-cu12 12.1.105
nvidia-cuda-runtime-cu12 12.1.105
nvidia-cudnn-cu12 9.1.0.70
nvidia-cufft-cu12 11.0.2.54
nvidia-curand-cu12 10.3.2.106
nvidia-cusolver-cu12 11.4.5.107
nvidia-cusparse-cu12 12.1.0.106
nvidia-nccl-cu12 2.20.5
nvidia-nvjitlink-cu12 12.1.105
nvidia-nvtx-cu12 12.1.105
omegaconf 2.1.2
opencv-python 4.10.0.84
opt_einsum 3.4.0
optree 0.13.0
packaging 24.1
pandas 1.4.1
peract_colab 0.0.0
pillow 10.4.0
pip 24.2
platformdirs 4.2.2
point_renderer 0.2.0
portalocker 2.10.1
proglog 0.1.10
protobuf 4.25.5
psutil 6.0.0
pycparser 2.21
pyglet 2.0.17
Pygments 2.18.0
PyOpenGL 3.1.0
pyparsing 3.1.4
pyquaternion 0.9.9
pyrender 0.1.45
PyRep 4.1.0.3
PySocks 1.7.1
python-dateutil 2.9.0.post0
pytorch3d 0.7.8
pytz 2024.2
PyYAML 6.0.2
regex 2024.9.11
requests 2.32.3
rich 13.9.1
rlbench 1.2.0
rvt 0.0.1
safetensors 0.4.5
scipy 1.14.1
setproctitle 1.3.3
setuptools 75.1.0
six 1.16.0
sympy 1.12
tabulate 0.9.0
tensorboard 2.17.1
tensorboard-data-server 0.7.2
tensorflow 2.17.0
tensorflow-io-gcs-filesystem 0.37.1
termcolor 2.4.0
timeout-decorator 0.5.0
tokenizers 0.20.0
tomli 2.0.1
torch 2.4.1+cu121
torchaudio 2.4.1+cu121
torchvision 0.19.1+cu121
tqdm 4.66.5
transformers 4.45.1
transforms3d 0.4.2
trimesh 4.4.9
triton 3.0.0
typeguard 4.3.0
typing_extensions 4.12.2
tzdata 2024.2
urllib3 2.2.3
wcwidth 0.2.13
Werkzeug 3.0.4
wheel 0.44.0
wrapt 1.16.0
xformers 0.0.28.post1
yacs 0.1.8
yarr 0.1
zipp 3.19.2

```

imankgoyal, 2024-10-04:
Hi,

Details are here: https://github.com/NVlabs/RVT/issues/45#issuecomment-2197729376

imankgoyal, 2024-10-14:
Closing because of inactivity. Feel free to open it if the issue persists.

closed 2024-10-18 by imankgoyal