NVlabs/RVT #63: RuntimeError: The call failed on the V-REP side. Return value: -1

LogSSim, 2024-10-14:
I deleted the code before, but when I downloaded the code again, I found it didn't work.
  File "/data3/urs/code/RVT/rvt/libs/RLBench/rlbench/backend/task.py", line 410, in _get_waypoints
    feasible, way_i = self._feasible(waypoints)
  File "/data3/urs/code/RVT/rvt/libs/RLBench/rlbench/backend/task.py", line 361, in _feasible
    path = point.get_path(ignore_collisions=True)
  File "/data3/urs/code/RVT/rvt/libs/RLBench/rlbench/backend/waypoints.py", line 55, in get_path
    path = arm.get_path(self._waypoint.get_position(),
  File "/data3/urs/code/RVT/rvt/libs/PyRep/pyrep/robots/arms/arm.py", line 444, in get_path
    p = self.get_nonlinear_path(
  File "/data3/urs/code/RVT/rvt/libs/PyRep/pyrep/robots/arms/arm.py", line 385, in get_nonlinear_path
    _, ret_floats, _, _ = utils.script_call(
  File "/data3/urs/code/RVT/rvt/libs/PyRep/pyrep/backend/utils.py", line 65, in script_call
    return sim.simExtCallScriptFunction(
  File "/data3/urs/code/RVT/rvt/libs/PyRep/pyrep/backend/sim.py", line 698, in simExtCallScriptFunction
    _check_return(ret)
  File "/data3/urs/code/RVT/rvt/libs/PyRep/pyrep/backend/sim.py", line 27, in _check_return
    raise RuntimeError(
RuntimeError: The call failed on the V-REP side. Return value: -1

imankgoyal, 2024-10-14:
Hi, Looks like some RLBench/V-REP related error. Maybe try the install again.

LogSSim, 2024-10-16:
OK, Thanks. I solved it.

codeshop715, 2024-10-16:
@LogSSim  I ran into the same problem. How did you resolve it?

LogSSim, 2024-10-17:
@codeshop715  I reinstalled COPPELIASIM and the following libraries：
pip install -e rvt/libs/PyRep 
pip install -e rvt/libs/RLBench 
pip install -e rvt/libs/YARR 
pip install -e rvt/libs/peract_colab
pip install -e rvt/libs/point-renderer

codeshop715, 2024-10-18:
@LogSSim  Thanks a lot.