NVlabs/RVT #65: num.of demonstrations used during training ?

rebecajohn, 2024-10-17:
Hi @imankgoyal , The model was trained using how many demonstrations for each task in RVT and RVT2?

rebecajohn, 2024-10-17:
I am curious to know why the performance in some tasks that was trained  on 10 demonstrations exceed than that of 100 demonstrations ?

imankgoyal, 2024-10-17:
Hi @rebecajohn, 

We use 100 demonstrations for each task in RLBench and 10 for each task in real. 100 was used in RLBench to keep the benchmark consistent with prior works. We haven't tested with 10 demonstrations in RLBench. Do you mean on some tasks, you get better results with 10 demonstrations in RLBench?

rebecajohn, 2024-10-18:
Hello @imankgoyal ,Yes. for example in RLBench , for RVT ,in Turn Tap task gives Avg success  for 10 demonstrations as 96 , but when the demonstrations are increased to 100 , Avg. success  decreases to 76 , even for some other tasks in those 18 tasks , we get small decrease in Avg. success.

closed 2024-10-19 by rebecajohn