NVlabs/RVT #67: Evaluation Always Fails

hanyangclarence, 2024-10-18:
Hi, thank you for your work!

I'm new in this field. I tried to run RVT 2 evaluation with visualizations on my device. I evaluated both push_button and stack_blocks tasks, and for both the model fails on all the episodes. When I watch the visualization video, I see that the robot arm just directly swings down in an abnormal manner, which doesn't seem like a well trained behavior. Also, I see that the success rate of push_button is 100% in the paper, so I guess the robot arm shouldn't behave in this way.

Hope you could tell me what could be wrong in my implementation, thanks!

hanyangclarence, 2024-10-18:
Another two observations:

1, in `models/rvt_agent.py` line 794
```
pc, img_feat = rvt_utils.move_pc_in_bound(
    pc, img_feat, self.scene_bounds, no_op=not self.move_pc_in_bound
)
```
I found that by default `self.move_pc_in_bound` is true, and this function returns empty pc and img_feat of length 0, meaning that all point clouds are filtered.

2, I also tried running rvt with
```
python eval.py --model-folder runs/rvt  --eval-datafolder ./data/test --tasks stack_blocks push_buttons --eval-episodes 2 --log-name test/1 --device 0 --headless --model-name model_14.pth  --save-video
```
This one results in an error:
```
Traceback (most recent call last):
  File "/home/hanyang/code/fyp/RVT/rvt/eval.py", line 556, in <module>
    _eval(args)
  File "/home/hanyang/code/fyp/RVT/rvt/eval.py", line 510, in _eval
    scores = eval(
  File "/home/hanyang/anaconda3/envs/rvt/lib/python3.9/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/home/hanyang/code/fyp/RVT/rvt/eval.py", line 300, in eval
    raise e
  File "/home/hanyang/code/fyp/RVT/rvt/eval.py", line 294, in eval
    for replay_transition in generator:
  File "/home/hanyang/code/fyp/RVT/rvt/libs/YARR/yarr/utils/rollout_generator.py", line 52, in generator
    act_result = agent.act(step_signal.value, prepped_data,
  File "/home/hanyang/anaconda3/envs/rvt/lib/python3.9/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/home/hanyang/code/fyp/RVT/rvt/models/rvt_agent.py", line 802, in act
    a, b = mvt_utils.place_pc_in_cube(
  File "/home/hanyang/code/fyp/RVT/rvt/mvt/utils.py", line 48, in place_pc_in_cube
    pc_mid = (torch.max(pc, 0)[0] + torch.min(pc, 0)[0]) / 2
IndexError: max(): Expected reduction dim 0 to have non-zero size.
```
It seems that the problem is similar to running rvt2.

imankgoyal, 2024-10-18:
Hi, Thanks for you interest in our work.

It seems to me that you are getting empty point clouds. There is likely some issue with your RLBench installation. I would suggest retrying to install RLBench and CoppeliaSim.

hanyangclarence, 2024-10-20:
> It seems to me that you are getting empty point clouds. There is likely some issue with your RLBench installation. I would suggest retrying to install RLBench and CoppeliaSim.

Thanks! My observation is that the point cloud and image feature are not empty before `models/rvt_agent.py` line 794. They are just all filtered out by that function. Could this be some issues related to RLBench installation? I installed RLBench with the official command `pip install -e rvt/libs/RLBench`, and can also successfully import it in the code. 

As for CoppeliaSim, I put the extracted folder to `/home/hanyang/CoppeliaSim_Player_V4_1_0_Ubuntu20_04` , and add the following to `~/.bashrc`:
```
export COPPELIASIM_ROOT="$HOME/CoppeliaSim_Player_V4_1_0_Ubuntu20_04"
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$COPPELIASIM_ROOT
export QT_QPA_PLATFORM_PLUGIN_PATH=$COPPELIASIM_ROOT
export DISPLAY=:0
```
One thing that is different from the guideline is the value of DISPLAY, yet if I set it to 1.0, it leads to the following error:
```
qt.qpa.xcb: could not connect to display :1.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/hanyang/CoppeliaSim_Player_V4_1_0_Ubuntu20_04" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, webgl, xcb.

Aborted
```
(I'm running on WSL2 ubuntu 20.04.6 LTS) So I follow the [guidelines](https://forum.qt.io/post/694990) to change the value to 0. But personally I don't think this setting could lead to empty point clouds?

imankgoyal, 2024-10-23:
I see. In my experience, even the point clouds getting filtered out had been caused by incorrect installation. One suggestion would be to visualize what the images and point clouds look like before the filtering.

hanyangclarence, 2024-10-25:
Thanks for the suggestion!

Another issue I found is that, although I can launch eval.py successfully, I cannot successfully run the example code in RLBench codebase. The error is like:
```
(rvt) hanyang@DESKTOP-6E066TJ:~/code/fyp/RLBench.arp/examples$ python few_shot_rl.py
Reset Episode

Error: signal 11:

/home/hanyang/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_Z11_segHandleri+0x30)[0x7fecddad6b50]
/lib/x86_64-linux-gnu/libc.so.6(+0x43090)[0x7fecebdbb090]
/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so(+0xd7deb7)[0x7fecbe2afeb7]
/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so(+0xd81994)[0x7fecbe2b3994]
/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so(+0xd79593)[0x7fecbe2ab593]
/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so(+0xd2395f)[0x7fecbe25595f]
/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so(+0xd2457e)[0x7fecbe25657e]
/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so(+0x2370ad)[0x7fecbd7690ad]
/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so(+0x237d23)[0x7fecbd769d23]
/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so(+0x2392ee)[0x7fecbd76b2ee]
QMutex: destroying locked mutex

Error: signal 11:

/home/hanyang/CoppeliaSim_Player_V4_1_0_Ubuntu20_04/libcoppeliaSim.so.1(_Z11_segHandleri+0x30)[0x7fecddad6b50]
/lib/x86_64-linux-gnu/libc.so.6(+0x43090)[0x7fecebdbb090]
/usr/lib/wsl/drivers/nvltig.inf_amd64_e6aa96348a13a85d/libnvwgf2umx.so(+0x2ca0a76)[0x7fecb0532a76]
/usr/lib/wsl/drivers/nvltig.inf_amd64_e6aa96348a13a85d/libnvwgf2umx.so(+0x2c9f69e)[0x7fecb053169e]
/usr/lib/wsl/drivers/nvltig.inf_amd64_e6aa96348a13a85d/libnvwgf2umx.so(+0x2c9f616)[0x7fecb0531616]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x8609)[0x7fecec0cc609]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x43)[0x7fecebe97353]
```

Does this happen to you?

imankgoyal, 2024-10-28:
Unfortunately, I have not tried running the example code from RLBench. I recommend first testing the train and eval code in the current repo.

hanyangclarence, 2024-10-30:
I see. I'll try to figure it out. Thanks!

hanyangclarence, 2024-11-11:
Thanks for you comments! Indeed it's my coppeliaSim's installation issues. Now it works correctly.

closed 2024-11-11 by hanyangclarence