NVlabs/RVT #69: How to Train and Test Using a Single Camera in RLBench?

LemonWade, 2024-10-23:
Thank you for your work! I have successfully reproduced your paper. However, I would like to know how to train and test using only one camera in RLBench. Any guidance would be greatly appreciated!

Thanks in advance!

imankgoyal, 2024-10-23:
It's good to hear that, @LemonWade. There is no pre-implemented way to do so; however, you can try modifying these parameters according to your needs and see if it works.

In Train: https://github.com/NVlabs/RVT/blob/master/rvt/models/rvt_agent.py#L295
In Eval: https://github.com/NVlabs/RVT/blob/master/rvt/eval.py#L165

LemonWade, 2024-10-25:
Thank you for your previous response. I have modified the `CAMERA` in the `rvt/utils/peract_utils.py` file to only include ["front"]. However, after running the training program, I encountered some errors related to replay. Below is the error message. 

```
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/root/anaconda3/envs/rvt2/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 302, in _worker_loop
    data = fetcher.fetch(index)
  File "/root/anaconda3/envs/rvt2/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 39, in fetch
    data = next(self.dataset_iter)
  File "/data/RVT/rvt/libs/YARR/yarr/replay_buffer/wrappers/pytorch_replay_buffer.py", line 39, in _generator
    yield self._replay_buffer.sample_transition_batch(pack_in_dict=True, distribution_mode = self._sample_distribution_mode)
  File "/data/RVT/rvt/libs/YARR/yarr/replay_buffer/uniform_replay_buffer.py", line 803, in sample_transition_batch
    store = self._get_from_disk(
  File "/data/RVT/rvt/libs/YARR/yarr/replay_buffer/uniform_replay_buffer.py", line 456, in _get_from_disk
    store[k][i] = v # NOTE: potential bug here, should % self._replay_capacity
KeyError: 'left_shoulder_rgb'
```

Could you please advise on how to proceed? Any suggestions would be greatly appreciated.

Thank you in advance!

closed 2024-10-28 by LemonWade