NVlabs/RVT #73: About eval error：“RuntimeError: Handle Panda does not exist”

LogSSim, 2024-11-21:
Why does this error occur?
"
RuntimeError: Handle Panda does not exist.
QMutex: destroying locked mutex
xvfb-run: error: problem while cleaning up temporary directory
"

LogSSim, 2024-11-21:
> Why does this error occur? " RuntimeError: Handle Panda does not exist. QMutex: destroying locked mutex xvfb-run: error: problem while cleaning up temporary directory "

Oh, I fix it

closed 2024-11-21 by imankgoyal

IrvingF7, 2025-06-24:
> > Why does this error occur? " RuntimeError: Handle Panda does not exist. QMutex: destroying locked mutex xvfb-run: error: problem while cleaning up temporary directory "
> 
> Oh, I fix it

Hi! Do you mind explaining how you fixed it? Thank you in advance