NVlabs/RVT #76: RuntimeError:CUDA Setup failed despite GPU being available. Please run the following command to get more information:

LogSSim, 2025-01-02:
Oh, God. When I changed the server environment and reinstalled rvt, I got an error:
'''
  File "/data2/s/code/RVT/rvt/models/rvt_agent.py", line 12, in <module>
    import bitsandbytes as bnb
  File "/home/s/anaconda3/envs/rvt/lib/python3.8/site-packages/bitsandbytes/__init__.py", line 6, in <module>
    from . import cuda_setup, utils, research
  File "/home/s/anaconda3/envs/rvt/lib/python3.8/site-packages/bitsandbytes/research/__init__.py", line 1, in <module>
    from . import nn
  File "/home/s/anaconda3/envs/rvt/lib/python3.8/site-packages/bitsandbytes/research/nn/__init__.py", line 1, in <module>
    from .modules import LinearFP8Mixed, LinearFP8Global
  File "/home/sjy/anaconda3/envs/rvt/lib/python3.8/site-packages/bitsandbytes/research/nn/modules.py", line 8, in <module>
    from bitsandbytes.optim import GlobalOptimManager
  File "/home/s/anaconda3/envs/rvt/lib/python3.8/site-packages/bitsandbytes/optim/__init__.py", line 6, in <module>
    from bitsandbytes.cextension import COMPILED_WITH_CUDA
  File "/home/s/anaconda3/envs/rvt/lib/python3.8/site-packages/bitsandbytes/cextension.py", line 20, in <module>
    raise RuntimeError('''
RuntimeError: 
        CUDA Setup failed despite GPU being available. Please run the following command to get more information:

        python -m bitsandbytes

        Inspect the output of the command and see if you can locate CUDA libraries. You might need to add them
        to your LD_LIBRARY_PATH. If you suspect a bug, please take the information from python -m bitsandbytes
        and open an issue at: https://github.com/TimDettmers/bitsandbytes/issues

'''

What is the reason for this?

imankgoyal, 2025-01-06:
Hi,

This seems like an error related to bitsandbytes. I have not encountered it in the past, so unfortunately, so unfortunately I can't help much with this. Maybe try posting your bits and bytes-related log in their official GitHub along with the version number.

violetdenim, 2025-01-22:
I solved this by performing
```
pip uninstall bitsandbytes –y
pip install bitsandbytes

```