NVlabs/RVT #86: qt.qpa.plugin: Could not find the Qt platform plugin "xcb"

ChunkHsu, 2025-02-22:
Thank you for your work!!! 
I am using a local machine with Ubuntu 20.04.
When I run `rvt/eval.py`, I get the following error (I checked the issue on https://github.com/NVlabs/RVT/issues/2, but regardless of whether I run it in `headless` mode or not, the error still occurs, and the machine has a local display):

```shell
(rvt) chunk@chunk-ubuntu:~/RVT/rvt$ python eval.py --model-folder runs/rvt2  --eval-datafolder ./data/val --tasks open_drawer --eval-episodes 25 --log-name test/1 --device 0 --headless --model-name model_99.pth
2025-02-22 17:09:21.584961: I tensorflow/core/util/port.cc:110] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-02-22 17:09:21.604017: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 AVX_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-02-22 17:09:21.857951: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
bin /home/chunk/miniconda3/envs/rvt/lib/python3.8/site-packages/bitsandbytes/libbitsandbytes_cuda111.so
WARNING: when three_views=True or two_views=False -- no_down=True no_top=False add_3p=False does not matter.
MVT Vars: {'training': True, '_parameters': OrderedDict(), '_buffers': OrderedDict(), '_non_persistent_buffers_set': set(), '_backward_hooks': OrderedDict(), '_is_full_backward_hook': None, '_forward_hooks': OrderedDict(), '_forward_pre_hooks': OrderedDict(), '_state_dict_hooks': OrderedDict(), '_load_state_dict_pre_hooks': OrderedDict(), '_load_state_dict_post_hooks': OrderedDict(), '_modules': OrderedDict(), 'depth': 8, 'img_feat_dim': 3, 'img_size': 224, 'add_proprio': True, 'proprio_dim': 4, 'add_lang': True, 'lang_dim': 512, 'lang_len': 77, 'im_channels': 64, 'img_patch_size': 14, 'final_dim': 64, 'attn_dropout': 0.1, 'decoder_dropout': 0.0, 'self_cross_ver': 1, 'add_corr': True, 'norm_corr': True, 'add_pixel_loc': True, 'add_depth': True, 'pe_fix': True, 'feat_ver': 1, 'wpt_img_aug': 0.0, 'inp_pre_pro': False, 'inp_pre_con': False, 'cvx_up': True, 'use_point_renderer': True, 'rot_ver': 1, 'num_rot': 72, 'no_feat': True}
MVT Vars: {'training': True, '_parameters': OrderedDict(), '_buffers': OrderedDict(), '_non_persistent_buffers_set': set(), '_backward_hooks': OrderedDict(), '_is_full_backward_hook': None, '_forward_hooks': OrderedDict(), '_forward_pre_hooks': OrderedDict(), '_state_dict_hooks': OrderedDict(), '_load_state_dict_pre_hooks': OrderedDict(), '_load_state_dict_post_hooks': OrderedDict(), '_modules': OrderedDict(), 'depth': 8, 'img_feat_dim': 3, 'img_size': 224, 'add_proprio': True, 'proprio_dim': 4, 'add_lang': True, 'lang_dim': 512, 'lang_len': 77, 'im_channels': 64, 'img_patch_size': 14, 'final_dim': 64, 'attn_dropout': 0.1, 'decoder_dropout': 0.0, 'self_cross_ver': 1, 'add_corr': True, 'norm_corr': True, 'add_pixel_loc': True, 'add_depth': True, 'pe_fix': True, 'feat_ver': 1, 'wpt_img_aug': 0.0, 'inp_pre_pro': False, 'inp_pre_con': False, 'cvx_up': True, 'use_point_renderer': True, 'rot_ver': 1, 'num_rot': 72, 'no_feat': False}
Using 8-Bit Optimizer
Agent Information
<rvt.models.rvt_agent.RVTAgent object at 0x7f73bec9fa60>
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in "/home/chunk/miniconda3/envs/rvt/lib/python3.8/site-packages/cv2/qt/plugins"
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Aborted (core dumped)
```

Additionally, I am unable to run CoppeliaSim individually. When I run `CoppeliaSim/coppeliaSim.sh`, I get the following error:

```shell
(rvt) chunk@chunk-ubuntu:~/CoppeliaSim$ ./coppeliaSim.sh 
[CoppeliaSimClient]    loading the CoppeliaSim library...
[CoppeliaSimClient]    done.
[CoppeliaSimClient:loadinfo]   launching CoppeliaSim...
[CoppeliaSim:loadinfo]   CoppeliaSim V4.1.0., (rev. 1), flavor: 1
[CoppeliaSim:loadinfo]   Legacy machine ID: 5000-9DEB-FFC4-9CEC-F7E3-841D
[CoppeliaSim:loadinfo]   Machine ID: 0E98-3B4E-77A1-0000-FA0E-0101
qt.qpa.xcb: could not connect to display :1.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/chunk/CoppeliaSim" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, webgl, xcb.

./coppeliaSim.sh: line 35: 16298 Aborted               (core dumped) "$dirname/$appname" "${PARAMETERS[@]}"
```
Prior to this, I had set up the environment for the [PerAct paper](https://github.com/peract/peract), and the `eval.py` from PerAct runs successfully, and I can also run `CoppeliaSim` individually. 

Last, this is the content of my `.bashrc` file:
```shell
...
# >>> RVT >>>
export COPPELIASIM_ROOT=/home/chunk/CoppeliaSim
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$COPPELIASIM_ROOT
export QT_QPA_PLATFORM_PLUGIN_PATH=$COPPELIASIM_ROOT
export DISPLAY=:1.0
export CUB_HOME=/home/chunk/cub-1.10.0
# <<< RVT <<<
```
These are the packages installed in the `rvt` environment:
```shell
Package                      Version      Editable project location
---------------------------- ------------ ---------------------------------------
absl-py                      2.1.0
antlr4-python3-runtime       4.9.3
astunparse                   1.6.3
bitsandbytes                 0.38.1
bypy                         1.8.5
cachetools                   5.5.2
certifi                      2022.12.7
cffi                         1.14.2
charset-normalizer           2.1.1
clip                         1.0
contourpy                    1.1.1
cycler                       0.12.1
decorator                    4.4.2
[... 72 lines cut ...]
tensorboard-data-server      0.7.2
tensorflow                   2.13.1
tensorflow-estimator         2.13.0
tensorflow-io-gcs-filesystem 0.34.0
termcolor                    2.4.0
timeout-decorator            0.5.0
tokenizers                   0.20.3
torch                        1.12.1+cu113
torchaudio                   0.12.1+cu113
torchvision                  0.13.1+cu113
tqdm                         4.67.1
transformers                 4.46.3
transforms3d                 0.4.2
trimesh                      4.6.3
typing_extensions            4.5.0
tzdata                       2025.1
urllib3                      1.26.13
wcwidth                      0.2.13
Werkzeug                     3.0.6
wheel                        0.44.0
wrapt                        1.17.2
yacs                         0.1.8
yarr                         0.1          /home/chunk/RVT/rvt/libs/YARR
zipp                         3.20.2
```

jeffrey-ke, 2025-03-14:
I had this same issue. You may need to apt install all the qt libraries to get all the plugins in places where rvt expects them.

xiezhipeng-git, 2025-03-16:
Similar errors
```
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/root/anaconda3/lib/python3.12/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, minimal, minimalegl, offscreen, vnc, webgl.

```
Try to solve it using the following methods.
```
conda install qt-main qt-webengine q
t-xcb
```
then bash.rc
```
export QT_QPA_PLATFORM_PLUGIN_PATH=/root/anaconda3/lib/python3.12/site-packages/cv2/qt/plugins/platforms
```
but it doesn't work

ChunkHsu, 2025-03-17:
Thanks for your response. I have resolved this issue by replacing the Qt plugin folder in my Conda environment with the QT5 plugin directory.

> I had this same issue. You may need to apt install all the qt libraries to get all the plugins in places where rvt expects them.

> Similar errors
>
> ```
> qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/root/anaconda3/lib/python3.12/site-packages/cv2/qt/plugins" even though it was found.
> This application failed to [...]

ChunkHsu, 2025-03-17:
> I had this same issue. You may need to apt install all the qt libraries to get all the plugins in places where rvt expects them.

Thanks for your response. I have resolved this issue by replacing the Qt plugin folder in my Conda environment with the QT5 plugin directory.

ChunkHsu, 2025-03-17:
> Similar errors
>
> ```
> qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/root/anaconda3/lib/python3.12/site-packages/cv2/qt/plugins" even though it was found.
> This application failed to [...]

I tried it, and it didn't work too.  : (